lvgl/lv_port_pc_eclipse #102: table memory overflow

wangyaoping, 2022-02-14:
Hi everyone:
 When I use table controls,and Create and delete repeatedly. The memory consumption of monitoring display is increasing（lv_sim_eclipse_sdl-release-v8.0). hope your answer
The code is as follows

`static lv_obj_t * tableMain=NULL;

void creat_obj_1()
{
    if(tableMain!=NULL)
        return;
    tableMain = lv_table_create(lv_scr_act());
}

void destroy_obj_1()
{
    if(tableMain!=NULL)
    {
        lv_obj_del(tableMain);
        tableMain=NULL;
    }
}

static void btn_event_handle(lv_event_t *e)
{
    unsigned int i=0;
    lv_obj_t * btn = lv_event_get_target(e);
    if(btn==btn_mem_test3)
    {
        for(i=0; i<1000; i++)
        {
            creat_obj_1();
            destroy_obj_1();
        }
    }
}`

kisvegabor, 2022-02-14:
I don't see any issue on my side.

Please attache a log where you can see the memory leak.

wangyaoping, 2022-02-14:
thinks, This problem has been solved through this web method<https://github.com/lvgl/lvgl/issues/2967>

closed 2022-02-14 by wangyaoping