lvgl/lv_port_pc_eclipse #109: build in windows using cygwin but no gui display

xiazp, 2022-07-28:
E:\project\lvgl\lv_port_pc_eclipse\cmake-build-debug\bin\main.exe
[Warn]	(0.000, +0)	 lv_init: Memory integrity checks are enabled via LV_USE_ASSERT_MEM_INTEGRITY which makes LVGL much slower 	(in lv_obj.c line #160)
[Warn]	(0.000, +0)	 lv_init: Object sanity checks are enabled via LV_USE_ASSERT_OBJ which makes LVGL much slower 	(in lv_obj.c line #164)
[Warn]	(0.000, +0)	 lv_init: Style sanity checks are enabled that uses more RAM 	(in lv_obj.c line #168)

kisvegabor, 2022-08-01:
Do you see at least a window opened (even without content)?

djcdei, 2023-09-09:
我也遇到这个问题

ArdaSenyurek, 2024-06-02:
I'm experiencing the same problem. I'm running on RPI4. I cloned lv_port_linux rep and built it as stated in its readme.md. I do sudo ./main inside bin folder but nothing shows up. Not even a window. I tried original main.c as well as some other lv_get_started functions inside the main loop but problem persists.

kisvegabor, 2024-06-07:
Do you have the UI (i.e. windows) of the Raspberry's OS running too? If so it overwrites what you draw into the frame buffer. 

If this is the case press `Ctrl + Alt + F1` to switch to command line mode and try running the LVGL app from there. 

More info [here](https://blog.lvgl.io/2018-01-03/linux_fb). If it's working I'll update the README.

loulansuiye, 2024-06-21:
> I'm experiencing the same problem. I'm running on RPI4. I cloned lv_port_linux rep and built it as stated in its readme.md. I do sudo ./main inside bin folder but nothing shows up. Not even a [...]

I experienced it to

kisvegabor, 2024-06-21:
Does [this](https://github.com/lvgl/lv_port_pc_eclipse/issues/109#issuecomment-2155412628) help?