lvgl/lv_port_pc_eclipse #114: MOUSE is not work on submodule lv_drivers v8.3.0

kay54068, 2022-10-20:
i want use the  lv_port_pc_eclipse project.

the demo is work fine on 8.3.0 dev(1bd4368e) , but i get some problems when i update to  lv_drivers to v8.3.0 (462d9cde):

[image]

i check is the lv_driver repo ,  the  issue is happened after 462d9cde.

btw,
the current  lv_drv_conf.h USE_SDL setting seems a bit strange:
[image]

kisvegabor, 2022-10-21:
I've updated both the `release/v8` and the `master` branch. All should work now.

kay54068, 2022-10-24:
@kisvegabor  
it's work for me.  thanks.

kisvegabor, 2022-10-24:
Great! :slightly_smiling_face:

closed 2022-10-24 by kisvegabor