lvgl/lv_port_pc_eclipse #117: ld: library not found for -lSDL2main

zhiping128, 2023-02-08:
when i run "lv_port_pc_eclipse" in eclipse IDE , have a error's

ld: library not found for -lSDL2main
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [lv_port_pc_eclipse] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

14:34:00 Build Failed. 2 errors, 0 warnings. (took 898ms)

zhiping128, 2023-02-09:
> Have you installed SDL as described in the [README](https://github.com/lvgl/lv_port_pc_eclipse#install-sdl)?

@kisvegabor 

Yes of course！
[image]

kisvegabor, 2023-02-10:
I see that you or Mac. Sorry, by literally don't know anything on developing on Mac :slightly_frowning_face: 

I suggest asking around the [Forum](https://forum.squareline.io/). And - as it doesn't seem like an LVGL specific issue - general Googling might help too.

closed 2025-05-19 by stale[bot]