lvgl/lv_port_pc_eclipse #118: needs update to be able to run on new lvgl master

uraich, 2023-03-06:
Does not compile any more because of changes to the SDL driver

kisvegabor, 2023-03-08:
Updated!

uraich, 2023-03-08:
There is still a problem:
I did:
git clone --recursive https://github.com/lvgl/lv_port_pc_eclipse.git
cd lv_port_pc_eclipse.git
mkdir build
cd build
cmake ..
and get this:
[image]

kisvegabor, 2023-03-08:
I've fixed the Cmake error. Noe looking into  lv_example_label_4.

kisvegabor, 2023-03-08:
I've fixed the example too.

It's workaround now because `lv_canvas` supports only native color formats (RGB) now. Later it should support any color format again.

uraich, 2023-03-08:
Thanks! 
Working now.

JAremko, 2023-11-05:
Annnd it broke again :)