lvgl/lv_port_pc_eclipse #121: unknown SDL_PixelFormatEnum

Y009, 2023-04-27:
On docker latest libsdl2-dev SDL_PixelFormatEnumdoes not seem to be available anymore

It can be fixed in /lvgl/src/dev/sdl/lv_sdl_window.c as following:
`
static void texture_resize(lv_disp_t * disp)
{
...
#if LV_COLOR_DEPTH == 32
    SDL_PixelFormat px_format = {.format = SDL_PIXELFORMAT_ARGB8888};
#elif LV_COLOR_DEPTH == 24
    SDL_PixelFormat px_format = {.format = SDL_PIXELFORMAT_BGR24};
#elif LV_COLOR_DEPTH == 16
    SDL_PixelFormat px_format = {.format = SDL_PIXELFORMAT_RGB565};
#else
#error("Unsupported color format")
#endif

    dsc->texture = SDL_CreateTexture(dsc->renderer, px_format.format,
                                     SDL_TEXTUREACCESS_STATIC, hor_res, ver_res);
    SDL_SetTextureBlendMode(dsc->texture, SDL_BLENDMODE_BLEND);
}`

Made a PR in the main repo [](https://github.com/lvgl/lvgl/pull/4172)

kisvegabor, 2023-04-28:
The PR is merged, thank you!

closed 2023-04-28 by kisvegabor

Y009, 2023-04-28:
I just checked with ubuntu 22.04 just in case and that pulls a newer version of libsdl (2.0.20) which does have SDL_PixelFormatEnum available, 
So I think it would actually be better to revert the commit and update the dockerfile to run with ubuntu 22.04 (or 23.04) instead.

kisvegabor, 2023-04-28:
Ubuntu 18.04 is really old. So yeah, updating the Docker file would be a more elegant solution. I'll revert the commit and updated the Docker in https://github.com/lvgl/lv_port_pc_eclipse/commit/25b91fd89277936c1199e083bf4e8201e4d11fdb

Y009, 2023-04-28:
Just checked on 23.04, it works there also

kisvegabor, 2023-04-28:
Okay, updated to 23.04. Thank you!