lvgl/lv_port_pc_eclipse #127: Release branches supported?

JPHutchins, 2023-08-28:
Hello, thanks for this great demo!

I'm curious if the simulator is intended to be supported on released versions of LVGL?

If this repo is modified to pull in a release version as dependency:

.gitmodules
```
[submodule "lvgl"]
	path = lvgl
	url = https://github.com/lvgl/lvgl.git
	branch = release/v8.2
```
```
git submodule update --remote
```

Then the build will fail due to missing include paths, definitions, etc.:
```
cmake -B build -G Ninja && cmake --build build --clean-first && ./bin/main

...

[7/399] Building C object CMakeFiles/main.dir/mouse_cursor_icon.c.o
FAILED: CMakeFiles/main.dir/mouse_cursor_icon.c.o 
/usr/bin/cc -DLV_CONF_INCLUDE_SIMPLE -DLV_LVGL_H_INCLUDE_SIMPLE -I/home/jp/repos/lv_port_pc_eclipse -I/usr/include/SDL2 -isystem /home/jp/repos/lv_port_pc_eclipse/lvgl -isystem /home/jp/repos/lv_port_pc_eclipse/lvgl/examples -isystem /home/jp/repos/lv_port_pc_eclipse/lvgl/demos -std=gnu11 -MD -MT CMakeFiles/main.dir/mouse_cursor_icon.c.o -MF CMakeFiles/main.dir/mouse_cursor_icon.c.o.d -o CMakeFiles/main.dir/mouse_cursor_icon.c.o -c /home/jp/repos/lv_port_pc_eclipse/mouse_cursor_icon.c
/home/jp/repos/lv_port_pc_eclipse/mouse_cursor_icon.c:49:16: error: ‘LV_COLOR_FORMAT_ARGB8888’ undeclared here (not in a function)
   49 |   .header.cf = LV_COLOR_FORMAT_ARGB8888,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
[24/399] Building C object CMakeFiles/main.dir/main.c.o
FAILED: CMakeFiles/main.dir/main.c.o 
/usr/bin/cc -DLV_CONF_INCLUDE_SIMPLE -DLV_LVGL_H_INCLUDE_SIMPLE -I/home/jp/repos/lv_port_pc_eclipse -I/usr/include/SDL2 -isystem /home/jp/repos/lv_port_pc_eclipse/lvgl -isystem /home/jp/repos/lv_port_pc_eclipse/lvgl/examples -isystem /home/jp/repos/lv_port_pc_eclipse/lvgl/demos -std=gnu11 -MD -MT CMakeFiles/main.dir/main.c.o -MF CMakeFiles/main.dir/main.c.o.d -o CMakeFiles/main.dir/main.c.o -c /home/jp/repos/lv_port_pc_eclipse/main.c
/home/jp/repos/lv_port_pc_eclipse/main.c:17:10: fatal error: lvgl/src/dev/sdl/lv_sdl_window.h: No such file or directory
   17 | #include "lvgl/src/dev/sdl/lv_sdl_window.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

My goal is to understand what support there is for simulator in the release branches.

Link to original forum thread: https://forum.lvgl.io/t/simulator-w-release-versions/12750

Cheers,
J.P.

kisvegabor, 2023-08-29:
Hi,

We have release branched is this repository:
[image]

You can do this:
```c
cd lv_port_pc_eclipse
git checkout release/v8.2
git submodule update --init
```

JPHutchins, 2023-08-29:
Cool!  Looks good to me!

closed 2023-08-29 by JPHutchins