lvgl/lv_port_pc_eclipse #129: problemn building pc_simulator

Kephas3, 2023-08-30:
Hi,

After reading and even following the tutorial in the LVGL course to install the simulator on Ubuntu 22.04.3 , i encountered the following issue.

[image]

It happens that, even after building a first time, the folder Debug doesn't  appear 
it finally appeared after creating the configuration in Run/Debug Settings, but the folder doesn't contain the pc_simulator_sdl_eclipse file.

Thank's for your help

kisvegabor, 2023-08-31:
Which version do you use? I've just pushed a fix to the `master` branch.

To solve it you need to remove `png` from project porpeties:
[image]

Kephas3, 2023-09-01:
Hello,

Thank's a lot it's working now :)

There is a bug that make the simulator crashed when navigating to Shop

```log
==17518==ERROR: AddressSanitizer: global-buffer-overflow on address 0x55915bd806a0 at pc 0x55915b9f0425 bp 0x7ffdf6d9c870 sp 0x7ffdf6d9c860
READ of size 1 at 0x55915bd806a0 thread T0
    #0 0x55915b9f0424 in rgb565_image_blend ../lvgl/src/draw/sw/blend/lv_draw_sw_blend_to_rgb565.c:265
    #1 0x55915b9ef935 in lv_draw_sw_blend_image_to_rgb565 ../lvgl/src/draw/sw/blend/lv_draw_sw_blend_to_rgb565.c:207
    #2 0x55915b9e1287 in lv_draw_sw_blend ../lvgl/src/draw/sw/blend/lv_draw_sw_blend.c:125
    #3 0x55915ba231a1 in lv_draw_sw_img ../lvgl/src/draw/sw/lv_draw_sw_img.c:212
    #4 0x55915ba029d6 in execute_drawing ../lvgl/src/draw/sw/lv_draw_sw.c:157
    #5 0x55915ba021d7 in lv_draw_sw_dispatch ../lvgl/src/draw/sw/lv_draw_sw.c:100
    #6 0x55915ba521f3 in lv_draw_dispatch_layer ../lvgl/src/draw/lv_draw.c:234
    #7 0x55915ba50932 in lv_draw_dispatch ../lvgl/src/draw/lv_draw.c:129
    #8 0x55915ba507dc in lv_draw_finalize_task_creation ../lvgl/src/draw/lv_draw.c:116
    #9 0x55915ba5aa7e in lv_draw_img ../lvgl/src/draw/lv_draw_img.c:92
    #10 0x55915b84438f in draw_img ../lvgl/src/widgets/img/lv_img.c:700
    #11 0x55915b840e3a in lv_img_event ../lvgl/src/widgets/img/lv_img.c:538
    #12 0x55915ba999ef in lv_obj_event_base ../lvgl/src/core/lv_obj_event.c:87
    #13 0x55915ba9b874 in event_send_core ../lvgl/src/core/lv_obj_event.c:307
    #14 0x55915ba995c2 in lv_obj_send_event ../lvgl/src/core/lv_obj_event.c:64
    #15 0x55915bad198d in lv_obj_redraw ../lvgl/src/core/lv_refr.c:104
    #16 0x55915badb74f in refr_obj ../lvgl/src/core/lv_refr.c:798
    #17 0x55915bad1eef in lv_obj_redraw ../lvgl/src/core/lv_refr.c:155
    #18 0x55915badb74f in refr_obj ../lvgl/src/core/lv_refr.c:798
    #19 0x55915bad1eef in lv_obj_redraw ../lvgl/src/core/lv_refr.c:155
    #20 0x55915badb74f in refr_obj ../lvgl/src/core/lv_refr.c:798
    #21 0x55915bad1eef in lv_obj_redraw ../lvgl/src/core/lv_refr.c:155
    #22 0x55915badb74f in refr_obj ../lvgl/src/core/lv_refr.c:798
    #23 0x55915bad1eef in lv_obj_redraw ../lvgl/src/core/lv_refr.c:155
    #24 0x55915badb74f in refr_obj ../lvgl/src/core/lv_refr.c:798
    #25 0x55915bad1eef in lv_obj_redraw ../lvgl/src/core/lv_refr.c:155
    #26 0x55915badb74f in refr_obj ../lvgl/src/core/lv_refr.c:798
    #27 0x55915badab96 in refr_obj_and_children ../lvgl/src/core/lv_refr.c:694
    #28 0x55915bada48f in refr_area_part ../lvgl/src/core/lv_refr.c:629
    #29 0x55915bad80d6 in refr_area ../lvgl/src/core/lv_refr.c:530
    #30 0x55915bad715a in refr_invalid_areas ../lvgl/src/core/lv_refr.c:494
    #31 0x55915bad4af8 in _lv_disp_refr_timer ../lvgl/src/core/lv_refr.c:371
    #32 0x55915b903f01 in lv_timer_exec ../lvgl/src/misc/lv_timer.c:343
    #33 0x55915b902417 in lv_timer_handler ../lvgl/src/misc/lv_timer.c:112
    #34 0x55915bb4b683 in main ../main.c:128
    #35 0x7f6856229d8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #36 0x7f6856229e3f in __libc_start_main_impl ../csu/libc-start.c:392
    #37 0x55915b7adb84 in _start (/home/pc/eclipse-workspace/pc_simulator_sdl_eclipse/Debug/pc_simulator_sdl_eclipse+0x45bb84)

0x55915bd806a0 is located 0 bytes to the right of global variable 'img_clothes_map' defined in '../lvgl/demos/widgets/assets/img_clothes.c:13:92' (0x55915bd7e1e0) of size 9408
SUMMARY: AddressSanitizer: global-buffer-overflow ../lvgl/src/draw/sw/blend/lv_draw_sw_blend_to_rgb565.c:265 in rgb565_image_blend
Shadow bytes around the buggy address:
  0x0ab2ab7a8080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a8090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a80a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a80b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a80c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0ab2ab7a80d0: 00 00 00 00[f9]f9 f9 f9 00 00 00 00 00 00 00 00
  0x0ab2ab7a80e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a80f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a8100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a8110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0ab2ab7a8120: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==17518==ABORTING
```

Also i 've tried first to run it on OSX 13.4.1, but an error showed me the following

```log
clang : error : the clang compiler does not support =pg option on version of OSX 10.9 and later
```

Best regards

kisvegabor, 2023-09-01:
I've just fixed it in LVGL :slightly_smiling_face: 
Please update the master branch.

gjvanderheiden, 2023-09-05:
Just did a new clone (git clone --recursive https://github.com/littlevgl/pc_simulator_sdl_eclipse.git), got this on mac os:

clang: error: the clang compiler does not support -pg option on versions of OS X 10.9 and later

gjvanderheiden, 2023-09-05:
If I try project properties-> C/C++ Build -> Tool Chain Editor -> Current toolchain: MacOSX GCC (instead of Cross GCC) is compiles.

CMake seams to work without editing, apart from 2 compile errors (just do what the compiler suggests) in LVGL itself.

Kephas3, 2023-09-05:
@gjvanderheiden I confirm, it also works for me 

Thank's a lot ! ;)