lvgl/lv_port_pc_eclipse #149: lv_port_pc_eclipse V9 build failure

19e2k67, 2024-02-02:
[image]
```
function 'Tvg_Result tvg_engine_init(Tvg_Engine, unsigned int)' definition is marked dllimportGCC
'Tvg_Result tvg_engine_init(Tvg_Engine, unsigned int)' redeclared without dllimport attribute: previous dllimport ignored [-Wattributes]GCC
Tvg_Result tvg_engine_init(Tvg_Engine engine_method, unsigned int threads)
Engine API
Initializes TVG engines. TVG requires the running-engine environment. TVG runs its own task-scheduler for parallelizing rendering tasks efficiently. You can indicate the number of threads, the count of which is designated threads. In the initialization step, TVG will generate/spawn the threads as set by threads count. tvg_engine_init(TVG_ENGINE_SW, 0); //Initialize software renderer and use the main thread only

Parameters:
engine_method – The engine types to initialize. This is relative to the Canvas types, in which it will be used. For multiple backends bitwise operation is allowed. - TVG_ENGINE_SW: CPU rasterizer - TVG_ENGINE_GL: OpenGL rasterizer (not supported yet)
threads – The number of additional threads used to perform rendering. Zero indicates only the main thread is to be used.

Returns:
Tvg_Result enumeration.

Return values:
TVG_RESULT_SUCCESS Succeed.
TVG_RESULT_FAILED_ALLOCATION An internal error possibly with memory allocation.
TVG_RESULT_INVALID_ARGUMENT Unknown engine type.
TVG_RESULT_NOT_SUPPORTED Unsupported engine type.
TVG_RESULT_UNKNOWN Other error.

Note:
The Initializer keeps track of the number of times it was called. Threads count is fixed at the first init() call.
```
I am using the version in the folder，
[image]
Low version (requires lv_drivers folder) can successfully build
[image]
Please tell me what I should do?

wxlinus, 2024-02-04:
I also encountered this problem
[image]

kisvegabor, 2024-02-05:
Does it help if you add `#define TVG_BUILD 1` [here](https://github.com/lvgl/lvgl/blob/master/src/libs/thorvg/thorvg.h#L29)?

19e2k67, 2024-02-05:
[image]

wxlinus, 2024-02-07:
I solved this problem by changing mingw64 to cygwin, and then I compiled and installed SDL2, but I couldn't use debug because gcc under win is missing ASAN related libraries

wxlinus, 2024-02-07:
[image]

> I solved this problem by changing mingw64 to cygwin, and then I compiled and installed SDL2, but I couldn't use debug because gcc under win is missing ASAN related libraries

kisvegabor, 2024-02-07:
Feel free to disable ASAN. It's just optional.

wendaoyifei, 2024-06-28:
I had the same problem, what's the meaning of changing mingw64 to cygwin @wxlinus .   I worded in windows platform and use mingw and vscode， I am not get used to use codeblock even thoug it's recommended in windows platfrom

19e2k67, 2024-06-29:
Should we ask questions in this [project](https://github.com/lvgl/lv_port_pc_vscode)?

kisvegabor, 2024-07-09:
> Should we ask questions in this [project](https://github.com/lvgl/lv_port_pc_vscode)?

Oh yes, as it's VS code please open a new issue in https://github.com/lvgl/lv_port_pc_vscode

closed 2024-07-09 by kisvegabor

yaoxuexill, 2024-07-15:
[image]
Adding a sentence with a red line may solve your problem