lvgl/lv_port_pc_eclipse #156: Linking Error on Windows using Clion

VanSharkholme, 2024-11-07:
Hi,
I'm trying to use the simulator on Windows with Clion, however when there's an error of  **undefined reference to 'SDL_main'** during linking. The steps I took are as follows: I cloned this project, then downloaded MinGW 14.2.0 and SDL2-2.30.9, then copied everything in the SDL2 x86_64-w64-mingw32 folder to the root folder of the MinGW, finally I configured the toolset in the Clion and tried to compile this project. The compilation completed with no errors but during linking I got the error mentioned above. Heres the output:

F:/MinGW/mingw64_lvgl/bin/../lib/gcc/x86_64-w64-mingw32/14.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: F:/MinGW/mingw64_lvgl/lib/libSDL2main.a(SDL_windows_main.o): in function `main_getcmdline':
/Users/valve/release/SDL2/SDL2-2.28.4-source/foo-x64/../src/main/windows/SDL_windows_main.c:80:(.text+0x152): undefined reference to `SDL_main'
collect2.exe: error: ld returned 1 exit status
mingw32-make[3]: *** [CMakeFiles\main.dir\build.make:123: F:/test/lvgl_sim/lv_port_pc_eclipse/bin/main.exe] Error 1
mingw32-make[2]: *** [CMakeFiles\Makefile2:109: CMakeFiles/main.dir/all] Error 2
mingw32-make[1]: *** [CMakeFiles\Makefile2:142: CMakeFiles/run.dir/rule] Error 2
mingw32-make: *** [Makefile:181: run] Error 2

Maybe there's something wrong when I'm installing SDL2? I don't know.
Thanks in advance for your help

VanSharkholme, 2024-11-08:
It turns out the problem is not solved, after changing the function name the main function that actually gets executed is the one in __main__.c, not the one in main.c

VanSharkholme, 2024-11-08:
I added "#include <SDL2/SDL.h>" in the main.c and now it works

kisvegabor, 2024-11-10:
It's very interesting. SDL.h is included [here](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_private.h#L20). 

Anyway, thank you for sharing the solution.

zyk-zyk, 2025-01-20:
hellp me
C:/Users/ZYK/Desktop/lvgl/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lasan: No such file or directory
C:/Users/ZYK/Desktop/lvgl/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/13.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lubsan: No such file or directory

kisvegabor, 2025-02-06:
You can disable the Sanitizers on these places:

[image]

[image]

[image]

closed 2025-06-24 by kisvegabor