lvgl/lv_port_pc_eclipse #159: Error in build: blend_helium.o: missing .note.GNU-stack section implies executable stack

symdeb, 2025-03-07:
Followed this:

https://docs.lvgl.io/8.3/get-started/platforms/pc-simulator.html

1. installed CDT
2. Git clone this repo
3.  Import into CDT
4. All compilations goes fine until  the end there is an error:

Error in Eclipse when building.

 `
Building target: lv_port_pc_eclipse_arch_driver
Invoking: Cross G++ Linker

warning: ./lvgl/src/draw/sw/blend/helium/lv_blend_helium.o: missing .note.GNU-stack section implies executable stack
/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
Finished building target: lv_port_pc_eclipse_arch_driver**
 

23:16:22 Build Failed. 1 errors, 1 warnings. (took 51s.667ms)
`

symdeb, 2025-03-07:
Went into the repo tree:
1. mkdir build
2. cd build
3. cmake ..
4. make
it builds, but I can only find the "main" demo. 
When running make in the root instead of in build folder, the same error as above occurs. 
It seems Eclipse need to be set to use cmake ?

kisvegabor, 2025-03-25:
Hi,

Sorry for the late reply. Which compiler do you use and what is it's version?